nasa/fprime #3553: Exclude state machine autocode from UT coverage

zimri-leisher, 2025-05-04:
| | |
|:---|:---|
|**_F´ Version_**| devel|
|**_Affected Component_**| State machines  |
---
## Problem Description

Exact same issue as https://github.com/nasa/fprime-tools/pull/245/files
Except needs to be applied to 

[image]
StateMachineAc.hpp/cpp files
## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.12.10-76061203-generic-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.24.2
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==3.0.0a8
Project submodules:
```

Tanuj-rai, 2025-05-06:
Hey @thomas-bc, can I work on this issue?

Tanuj-rai, 2025-05-07:
Thanks for confirming! @thomas-bc, I've made the change in gcovr.py as suggested and opened a PR in fprime-tools to address the issue. Please let me know if there is anything else that needs to be done.

closed 2025-05-08 by LeStarch