nasa/fprime #3555: Make grid cards on the website selectable

SiboVG, 2025-05-04:
When going to sidebar pages on the website documentation, for example [User Manual](https://fprime.jpl.nasa.gov/v3.6.2/docs/user-manual/), you are greeted with a nice overview of the different chapters in grid cards. When I first saw this, I tried to click on a card to go to the corresponding page, but nothing happened. There even is a hover animation over the cards, leading me to further believe that I could interact with it.

It would be a nice enhancement if the cards were a hyperlink to the page they are referring to. Clicking the card brings you to the right page.

Some cards already contain a button to go to the page, e.g. "Visit Tutorial". Maybe those should be removed if the entire card is clickable. Or you can keep them if you wish to emphasize user engagement.

thomas-bc, 2025-05-06:
@Tanuj-rai thanks for the help - someone is looking into this already. Please feel free to take any issue that is tagged "Easy First Issue" or "Help Wanted" if you would like to contribute! :)

Tanuj-rai, 2025-05-06:
No problem @thomas-bc. Thank you for the update.

thomas-bc, 2025-05-08:
Closing this. Grid cards do not have a hover animation anymore. Making them clickable was not desirable since we're trying to stick with Markdown as much as possible.

closed 2025-05-08 by thomas-bc