nasa/fprime #3558: Markdown rendering on the web site flattens hierarchical lists into a single list

bocchino, 2025-05-06:
For example, on this page

https://fprime.jpl.nasa.gov/latest/docs/user-manual/framework/data-products/#2-basic-concepts

in Section 2.2, the list items are numbered 1 through 6, when they should be 1 and 2.a through 2.d. Compare the GitHub Markdown rendering on this page:

https://github.com/nasa/fprime/blob/devel/docs/user-manual/framework/data-products.md

This flattening makes the structure hard to understand.

bocchino, 2025-05-06:
Note: In Section 3.4.2 on that page, it looks like the second level of hierarchy is flattened, but the third level is retained.

thomas-bc, 2025-05-06:
https://github.com/Python-Markdown/markdown/issues/364
https://github.com/Python-Markdown/markdown/issues/3
https://github.com/mkdocs/mkdocs/issues/545

Linking in some issues that _might_ be related. There's likely a bug somewhere. We do have 4 space of indent so I would expect this to work, but maybe the newlines in between make it unhappy, or something else.

thomas-bc, 2025-05-06:
@ashleynilo would you be able to take a look ?
I suspect the extra newlines in between each element might be the issue. We should try to convert
```md
   1. Text A

      1. Text B

      1. Text C
```
into
```md
   1. Text A
      1. Text B
      1. Text C
```

and see if that fixes it. If not we can look into it more, or ask the Python-Markdown folks if they know

bocchino, 2025-05-06:
It could also be that the source text uses 3-space indentation, that GitHub renders this indentation as hierarchy, and that the web site rendering requires 4 spaces.

thomas-bc, 2025-05-06:
Ohh yes, somehow I counted 4 but it looks like it's 3-space. We should switch to 4-space indentation

bocchino, 2025-05-06:
Let's first confirm that's the issue. If so, there is another issue to solve: since we review and accept these docs via GitHub, and the web site renders them differently, how can we write these docs in such a way that they will be rendered as expected on the web site? Can we render the docs way the web site will render them at the time that they are reviewed and accepted? Or can we run a script to find and exclude 3-space indentation and other inconsistencies between GitHub rendering and web rendering? Otherwise these errors can be sprinkled around the docs, and they are hard to detect.

thomas-bc, 2025-05-06:
Mkdocs uses https://github.com/Python-Markdown/markdown to render markdown into HTML. GitHub uses a different engine. 

The website is not super straightforward to build locally, since we also reference content that is version-controlled within JPL. Not impossible to just render the external part, but we'll need to put out instructions, or better some kind of script.

bocchino, 2025-05-06:
Also, I suspect there may be other discrepancies between the two tools. GitHub uses its own "flavor" of Markdown. It seems like mkdocs uses a different "flavor." The "flavors" seem to be mostly, but not entirely compatible, so in general we shouldn't expect anything to look a certain way on the web site because it looks that way in GitHub.

thomas-bc, 2026-01-13:
Fixed in devel https://fprime.jpl.nasa.gov/devel/docs/user-manual/framework/data-products/

closed 2026-01-13 by thomas-bc