nasa/fprime #3561: Support changing fprime-gds log level

celskeggs, 2025-05-06:
| | |
|:---|:---|
|**_F´ Version_**| v3.6.0ish |
|**_Affected Component_**| fprime-gds |
---
## Feature Description

Support a `--log-level` option that can be used to change the log level for fprime-gds.

## Rationale

This would be very helpful when debugging an issue that results in DEBUG logs. The default is to only save INFO logs and above.

Tanuj-rai, 2025-05-09:
[devel/src/fprime_gds/executables/cli.py
Lines 708 to 715 #](https://github.com/nasa/fprime-gds/blob/devel/src/fprime_gds/executables/cli.py#L650-L671)

Tanuj-rai, 2025-05-09:
Hey @celskeggs, I am interested in this issue. Can I work on it?

celskeggs, 2025-05-09:
That's a good question for @thomas-bc. I'm just reporting the issue.