nasa/fprime #3564: Remove FW_BAREMETAL_SCHEDULER from FpConfig.h

celskeggs, 2025-05-07:
These lines are still present in FpConfig.h:

```
#ifndef FW_BAREMETAL_SCHEDULER
#define FW_BAREMETAL_SCHEDULER \
    0  //!< Indicates whether or not a baremetal scheduler should be used. Alternatively the Os scheduler is used.
#endif
```

However, it appears that FW_BAREMETAL_SCHEDULER is no longer used anywhere, and that other mechanisms are now in place for the behavior that it previously offered.

Let's remove this unused setting so that it doesn't result in any confusion.

Tanuj-rai, 2025-05-08:
Hey @celskeggs, I’m interested in working on this issue. Can you please assign it to me?

celskeggs, 2025-05-08:
I'm not on the F Prime core team. @LeStarch?

thomas-bc, 2025-05-08:
@Tanuj-rai feel free to just open a PR

LeStarch, 2025-05-08:
Yeah, this is already in work on the cmake config pr.

LeStarch, 2025-05-08:
Will be fixed by #3563

LeStarch, 2025-05-08:
@Tanuj-rai sorry, I had this work done before the issue was reported and it is already in a PR.  I appreciate your desire to help.

closed 2025-05-10 by LeStarch

Tanuj-rai, 2025-05-11:
@LeStarch, No worries at all, thanks for letting me know. I'm glad it's already being handled! Apologies for the late reply.