nasa/fprime #3577: ROSES - Student review of Third-Party Library integration

thomas-bc, 2025-05-09:
### Description

Student Review of https://github.com/nasa/fprime/issues/3056

Docs and examples:
- https://fprime.jpl.nasa.gov/devel/docs/how-to/integrate-external-libraries/
- https://github.com/nasa/fprime-examples/tree/devel/FlightExamples/ExternalLibs

gorsheep, 2025-06-24:
**Approach 1 (CMake library with add_subdirectory)**

- The summary table currently does not mention any drawbacks. If there are truly none, it might be helpful to explicitly indicate this by writing “None” or “N/A.”
- The “Considerations” row describes this as the “Recommended approach if available,” but it would be useful to elaborate on why this is the case. Including a brief explanation of the advantages that make this the preferred method would provide helpful context.

gorsheep, 2025-06-24:
**Approach 2 (CMake library with FetchContent)**

- Similar to Approach 1, the summary table does not list any potential drawbacks. Adding this information, if applicable, would give readers a more balanced view.
- The documentation currently lists identical benefits and considerations for both Approach 1 and Approach 2. However, a direct comparison of the two would be valuable. For example, Approach 1 (cloning the library once and integrating it into the repo) ensures a stable, known version. In contrast, Approach 2 involves cloning a third-party repository during configuration, which could result in unintentional version changes if the upstream library is updated. Highlighting such distinctions would aid users in making an informed decision.

gorsheep, 2025-06-24:
**Use of Examples**

The use of concrete examples throughout the documentation is excellent. They effectively illustrate both integration strategies as well as important patterns such as the wrapper pattern, making the concepts more accessible and practical.

gorsheep, 2025-06-24:
**Clarification of the Term “Module”**

The documentation frequently uses the term _“module,”_ but this terminology may not align with standard F′ vocabulary, which typically refers to _“projects”_ and _“components.”_ It would be beneficial to clarify what is meant by “module” in this context. If the term is being used to refer to any directory with a `CMakeLists.txt` file, consistent with general CMake usage rather than F′ conventions, this distinction should be made explicit. This is especially important when providing guidance such as: _“This code can be located in the root project.cmake file if the library is used by multiple modules or directly in the module's CMakeLists.txt if it is only used by that module.”_ Without a clear definition of “module,” it may be unclear to readers which `CMakeLists.txt` file they are expected to modify. Clarifying this would help prevent confusion during integration.

gorsheep, 2025-06-24:
**Suggestion for a “Common Issues” Section**

Including a section on common pitfalls and troubleshooting tips would add practical value to the documentation. For instance, one common issue when integrating third-party libraries is name collisions between CMake targets, especially when a library brings in additional dependencies. Providing a brief warning about this, along with a recommended mitigation strategy, such as consistently applying namespaces to custom targets, would be highly useful for developers and reduce friction during integration.

closed 2025-06-26 by thomas-bc