nasa/fprime #3598: ComQueue hits an ASSERT if used without one of the queues

celskeggs, 2025-05-13:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| Svc.ComQueue |
---
## Problem Description

I'm working with a deployment that does not use file downlink. Therefore, I did not configure `configurationTable.entries[2]` when calling `comQueue.configure`. However, this resulted in `depth` being 0, which ultimately caused the following assertion:

```
Assert: "/workspaces/[...]/lib/fprime/Utils/Types/Queue.cpp:61" 0
```

This assertion was tripped when `ComQueue` tried to execute this line in `ComQueue::processQueue`:

```
if (queue.getQueueSize() == 0) {
```

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.18.0-553.47.1.el8_10.x86_64-x86_64-with-glibc2.34
Python version: 3.9.21
CMake version: 3.24.2
Pip version: 21.3.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==3.0.0a8
Project submodules:
    https://github.com/nasa/fprime.git @ v3.6.0-92-g6be3d3a9b
    https://github.com/fprime-community/fprime-baremetal @ 162b7dd
```

## How to Reproduce

See above.

## Expected Behavior

ComQueue should function correctly even if one of its features is not required, and should not require allocating memory that will not be used, solely for the purpose of not crashing.

thomas-bc, 2025-05-13:
@celskeggs did you also modify the ComQueue ports constants here ?

https://github.com/nasa/fprime/blob/ae2940b6c0cff2fe7138f581f56b90cb896f60ce/default/config/AcConstants.fpp#L36-L40

thomas-bc, 2025-05-13:
Ah, from FPP User Guide

> each named port instance is actually an array of one or more port instances

So yes ComQueue likely isn't usable in its current form without one of the queue

celskeggs, 2025-05-14:
Yes, I tried that. Setting ComQueueBufferPorts to 0 caused an FPP error, as you predicted.

LeStarch, 2025-05-14:
I don't know that we'll easily overcome the zero-port-array issue.  Be we certainly could allow configurations to set queue depths to 0, or otherwise specifically disable that particular queue.

closed 2026-09-10 by LeStarch