nasa/fprime #3601: ComStub and ComQueue expectations for ByteStreamModel implementations is not properly documented

celskeggs, 2025-05-14:
I have an interface implementing ByteStreamModel that cannot always accept more data to send due to limited bandwidth. I cannot return `ByteStreamStatus::SEND_RETRY` from my send port, because then ComStub will simply retry repeatedly, which will fail, wasting CPU cycles. So I return `ByteStreamStatus::OTHER_ERROR`.

But when I return an error, ComStub and ComQueue agree to stop sending me any data until I invoke the `ByteStreamReady` port again. This expectation is not documented in `ByteStreamDriverModel.fpp` nor `ByteStreamDriverModel/docs/sdd.md`. I request for it to be documented.

celskeggs, 2025-05-14:
The problem is that I _wasn't_ implementing the Communication Adapter interface. That was implemented for me by ComStub. I was implementing the ByteStreamDriver interface. ByteStreamDrivers are not solely used with ComStub; they can be used in other circumstances.

So it's not just about where the documentation is located, but also what it's documentation _about_.

Khan-Tzar, 2026-10-28:
Opened #6276 for this. It adds a "Ready" section to the ByteStreamDriverModel SDD: ComStub stops sending after OTHER_ERROR (or SEND_RETRY past its retry limit) until the driver invokes ready again, SEND_RETRY is retried immediately so it isn't a back-pressure mechanism, and the ready port annotations say the same. @celskeggs does that cover what you were missing?

closed 2026-10-28 by LeStarch