nasa/fprime #3605: Support overriding the time base in the F Prime GDS

celskeggs, 2025-05-14:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**| GDS |
---
## Feature Description

I'm using a target device that doesn't have a RTC, so I am not able to log channel values using wall clock time. Instead, I use the Svc.OsTime component with Os.RawTime. This means that the GDS sees the times as being on 1970-01-01. This causes problems with graphing.

I request that we either:

1. Support a configuration option to change the time base using a linear shift, OR
2. Add plugin support for providing custom interpretation of times.

## Rationale

Better support for embedded devices without RTCs.

timcanham, 2025-05-14:
@celskeggs You can use a time base of TB_PROC_TIME:

https://github.com/nasa/fprime/blob/devel/default/config/FpConfig.h#L31

The GDS will just display the time as the upper and lower raw values instead of 1970+. I do agree with providing for plug-ins based on the time base.

celskeggs, 2025-05-14:
@timcanham Are you _sure_ this is rendered correctly by the GDS? My testbed shift just ended, so I can't check again, but I'm pretty sure I was seeing the 1970+ date, and I _was_ using TB_PROC_TIME.

timcanham, 2025-05-14:
@celskeggs I admit I'm going from memory, but that's what I remembered. I'll check when I get a chance.

celskeggs, 2025-05-14:
Just checked the data lingering in my GDS tab. For a particular channel, I could see:

`_datastore.channels[0x4601].time.base` is `1` (TB_PROC_TIME)
`_datastore.channels[0x4601].time.seconds` is `58`
`_datastore.channels[0x4601].time.microseconds` is `608490`

And it was displayed like this:

[image]

Now, it _was_ correctly rendered in the channel log:

`TB_PROC_TIME: 58.608490s`

And that makes sense, since the `timeToString` implementation does not consider TB_PROC_TIME:
https://github.com/nasa/fprime-gds/blob/ba05c6d2059506cfd248292026010136aff15c1b/src/fprime_gds/flask/static/js/vue-support/utils.js#L100-L125

timcanham, 2025-05-14:
I stand corrected then... I wonder if the handling has changed.

closed 2025-12-11 by thomas-bc