nasa/fprime #3615: FpySequencer V0.3

zimri-leisher, 2025-05-14:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

A quick description of the requested feature.

## Rationale

A rationale for why the feature should be implemented in F´.

zimri-leisher, 2025-05-14:
PolyType for lvars...
Byte array somewhere
1 byte directive opcode, cmd is just a directive
promote all integer/float types to a higher level type?

zimri-leisher, 2025-05-14:
assert if not connected to prm/tlm db

zimri-leisher, 2025-05-19:
really want to avoid rewriting the AST. you will lose information about it

* should local variables have scope?
* start a document detailing the design of the high and low language
* Rob wants a formal grammar definition. start with https://docs.python.org/3/reference/grammar.html
   * maybe can generate a parser for the cut down grammar
* we likely need a new set of updated requirements
   * do we want requirements for the high level language?
   * or just a spec?
* idea: **one requirement, that we are required to implement the spec**
* only have F, U and I registers. also everything is LVARs-- byte arrays are "complex" lvars
* think about only having U, cuz u shouldn't need I cuz the arithmetic works the same way

memory in the sequencer for the sequence that gets reset when sequence is over is an lvar
* complex lvars (c0, c1, c2) are byte arrays
* simple lvars are integer/floating point 64 bit values (f0, f1, i0, i1)

zimri-leisher, 2025-05-28:
TODO add a local seq run port to fpyseq

zimri-leisher, 2025-06-04:
Considering renaming "local vars" to local buffers, and "registers" to local vars.
Local buffers = a byte array which tlm/prms go into
Local vars = 8 byte long "registers" used for math/logic ops

zimri-leisher, 2025-06-05:
In the meantime, I am just forging ahead as best I can. I've implemented the following binary comparison operator directives:
```
            EQ = 16
            NE = 17
            ULT = 18
            ULE = 19
            UGT = 20
            UGE = 21
            SLT = 22
            SLE = 23
            SGT = 24
            SGE = 25
```
I'm following the LLVM IR approach of having separate comparison inequality ops for signed and unsigned. I believe this should be the right approach.

zimri-leisher, 2025-06-09:
* write down all primitive types
  * don't forget strings
* local vars are USUALLY stack allocated, so don't be confusing about it
* maybe lvars are just reserved memory?
* remove "local vars", use all stack

zimri-leisher, 2025-06-09:
* separate createsymboltable from checking that the type exists

zimri-leisher, 2025-06-11:
@bocchino can you send the link to the list of compiler passes in fpp again? I lost it and not sure where to find it.

bocchino, 2025-06-11:
Sure! Here is the design:

https://github.com/nasa/fpp/wiki/Checking-Semantics

Here is the Scala implementation:

https://github.com/nasa/fpp/blob/main/compiler/lib/src/main/scala/analysis/CheckSemantics/CheckSemantics.scala

zimri-leisher, 2025-06-11:
Thanks @bocchino . Another question: often, I find myself deciding between a more permissive frontend (putting more work on the backend) or a more restrictive frontend (rejecting more things at the lexer/parser level). Are there any general principles on where's best to put this work? I can provide more specific examples.

zimri-leisher, 2025-06-16:
talked in meeting, good principle to simplify grammar--make more uniform

zimri-leisher, 2025-06-19:
@bocchino  thoughts on equality between two different types which have the same memory layout? E.G. comparing:

```
struct A {
    x: U32
}
```

and
```
struct B {
    y: U32
```

If I compare the two, I think they should be unequal even if x == y. Do you agree?

Kronos3, 2025-06-19:
@zimri-leisher Shouldn't this be disallowed at compile time?

zimri-leisher, 2025-06-19:
Good point, it's most likely a coding error. I will disallow it.

zimri-leisher, 2025-06-19:
Some todos:
* write up in spec sheet behavior of unsigned vs signed comparisons, weird edge cases
* short circuit or/ands

zimri-leisher, 2025-06-20:
It's working!!!!!!!!!!!!!
Input sequence:

[image]
Output bytecode:
[image]

There's lots of low hanging fruit for bytecode optimizations, but I'm going to focus on stability and cleaning up the code first. I am so happy! I'll run this in the sequencer tomorrow to prove it's actually usable.

zimri-leisher, 2025-06-20:
question:
what should "quit the sequence with a positive result" and "quit the sequence with a negative result" be called in the high level language?
positive result: `stop()` maybe?
negative result: `abort()` maybe?

or I could make it one function, just called `exit` and if it's `exit(False)` then it is failure, `exit(True)` is success

zimri-leisher, 2025-06-23:
@bocchino I will transition the bytecode to be stack-based, but I'm not going to do it this update, I've decided. I'll push (haha) it to the next release, and that will give me some more time to figure out which set of directives I'll need to implement.
For accessing serializables like tlm/prms, I will use the stack as allocated memory. For operations, I will use the stack as a workspace in the standard way, similar to how java does it.
Some questions:
Do I need a push_1/2/4/8/16 directive? or is just push_8 fine? likewise for pop.
Also, when using the stack as just "allocated memory", I guess I'll need to have something like
```
push_from_offset(stack offset, size)
```
which would read `size` bytes from the stack at `stack offset` and push them onto the top of the stack.

closed 2025-08-20 by zimri-leisher