nasa/fprime #3628: fprime-gds: Have channel/event template return format string with format specifiers filled in

timcanham, 2025-05-19:
| | |
|:---|:---|
|**_F´ Version_**|fprime-gds [v4.0.0a1](https://github.com/nasa/fprime-gds/releases/tag/v4.0.0a1) |
|**_Affected Component_**| n/a |
---
## Feature Description

Have the template `get_format_str()` functions return a format string with the FPP-inferred format specifiers filled in.

## Rationale

If the current format strings with "text {} blah" are supplied to an external user, there is no way to know what format specifiers to use. If FPP does automatic mapping, it should supply the format specifiers it inferred.

thomas-bc, 2025-06-06:
If the object is an `EventTemplate`/`ChTemplate` ([here](https://github.com/nasa/fprime-gds/blob/devel/src/fprime_gds/common/templates/ch_template.py#L19-L20)), they are not associated with a particular sample of telemetry, so there's no value to populate.
If the object is an `EventData`/`ChData`, then there is a [get_display_text()](https://github.com/nasa/fprime-gds/blob/devel/src/fprime_gds/common/data_types/ch_data.py#L117) that will do this.

Is this what you're looking for?

thomas-bc, 2025-06-06:
--> provide another `get_typed_format_str()` or something that inserts the C-style format specifiers in the tokens

closed 2025-12-19 by thomas-bc