nasa/fprime #3663: ROSES - Student review of Custom Framing documentation and example

thomas-bc, 2025-05-28:
### Description

https://github.com/nasa/fprime/issues/3633 was implemented here:

- https://github.com/nasa/fprime/pull/3635
- https://github.com/nasa/fprime-examples/pull/8

Needs student review

thomas-bc, 2025-07-29:
Note:
need to fix the part where it says to include the `.fppi` file to implement the interface https://fprime.jpl.nasa.gov/devel/docs/how-to/custom-framing/#flight-software-implementation
Now should be `import Svc.Framer`

gorsheep, 2025-10-15:
### 2. Expand the Discussion of When a FrameDetector Is Needed
The document briefly mentions that a `FrameDetector` is needed for stream-based transports like TCP or UART. It would be helpful to include additional examples relevant to real-world deployments, for example, whether **ZMQ** (the default transport used by the F´ GDS) requires a frame detector.  
Clarifying these distinctions would make it easier for users to determine whether a `FrameDetector` is necessary for their setup.

gorsheep, 2025-10-15:
### 3. Distinguish Between Downlink and Uplink Requirements
It would be useful to explain that:
- A **framer** is required for downlink (flight software → GDS).
- A **deframer** and **FrameDetector** are only needed if uplinking (GDS → flight software).  

Many users initially focus on downlink functionality; clarifying this distinction would prevent unnecessary implementation effort for one-way use cases.

gorsheep, 2025-10-15:
### 11. Mention Endianness Conventions in F´ Serialization
It would be valuable to remind readers that **F´ serializes all integer types in big-endian (network byte order)**, as defined in its core `Fw::Serializable` implementation. If a custom protocol requires little-endian fields, the user must explicitly swap bytes when building or parsing frames. This clarification would help prevent subtle interoperability bugs.

closed 2025-10-16 by LeStarch