nasa/fprime #3676: Custom Dictionaries

zimri-leisher, 2025-06-01:
| | |
|:---|:---|
|**_F´ Version_**| 4.0.0|
|**_Affected Component_**| json dictionary |
---
## Feature Description

Decisions as to what should/should not be in the dictionary should be left to the user, with a sensible default. I've noticed that we're trending towards adding more and more stuff to the dict. At some point, this will just approach putting the entire output of fpp-to-json into the dict. Maybe the user should be able to choose how their dict gets constructed, as a set of filters on the fpp-to-json output.

kevin-f-ortega, 2025-07-18:
Let's make this a discussion with the broader F Prime team.

thomas-bc, 2025-08-20:
Will be addressed by https://github.com/nasa/fpp/issues/762

Closing this one as duplicate

closed 2025-08-20 by thomas-bc