nasa/fprime #3688: Execution time limit annotations

celskeggs, 2025-06-03:
| | |
|:---|:---|
|**_F´ Version_**| n/a |
|**_Affected Component_**| FPP |
---
## Feature Description

For a project that I'm working on, it would be very useful to be able to define execution time limits on specific port calls. For example:

```
    async input port transmit: Fw.Buffer duration 50us    # handling "transmit" messages should take no more than 50us
    sync input port schedIn: Svc.Sched duration 200us     # executing the rategroup behaviors for this component should take no more than 200us
    output port loadDmaEngine: Fw.Buffer duration 10us    # the component that handles this port call should take no more than 10 us to do so
```

The details of how this is enforced (or whether it is only measured) would presumably be platform-specific, and don't need to be implemented in the F Prime core. But it would still require some level of FPP modeling support and autocoding support.

Since this might be a niche feature, it would also satisfy my needs to provide a generic mechanism that can be used to implement this feature on a platform-specific or project-specific basis. For example, I understand that FPP has an existing "annotations" feature that can be used to add additional unstructured information to FPP models. However, there would also need to be a generic feature that allows for the required code generation.

## Rationale

Some control loops involve executing code across many components as part of critical, hard-real-time behaviors. It would be useful to be able to model these deadlines and inject code to enforce them on a granular basis, rather than relying on higher-level analysis and testing.

zimri-leisher, 2025-06-03:
This would be amazing. I've found myself manually annotating ports with their expected maximum execution durations. If these were native FPrime features you could do some extremely powerful compile-time guarantees about execution time.

kevin-f-ortega, 2025-07-18:
It's not clear how useful this will be inside of FPP. Having a prototype would help to understand the usefulness of this feature. This is something that is not in the F Prime's roadmap; but, we are interested in external contributors to work on this.

As of now, annotations already show in the fpp-to-json files.

nikhilred2619, 2026-10-05:
Hi maintainers, I’d like to work on this issue. I’m considering an FPP-level mechanism for specifying execution-time constraints on port calls, with code-generation hooks that allow platform-specific measurement or enforcement while keeping the F´ core platform-independent. I would also include parser/model validation and regression tests. Is this direction aligned with the maintainers’ expectations, and would you be comfortable with me preparing a design/PR?