nasa/fprime #3713: Support for bulk telemetry

celskeggs, 2025-06-10:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**| FPP |
---
## Feature Description

FPP currently only allows specifying telemetry channels one at a time. For example:

```
        telemetry DeviceXRegisterA: U16
        telemetry DeviceXRegisterB: U16
        telemetry DeviceYRegisterA: U16
        telemetry DeviceYRegisterC: U16
```

However, there are some software systems that need hundreds or thousands of telemetry items to be produced by a component, driven by a table. While sometimes there is a way to use an array or a struct to summarize many telemetry channels into fewer channels, sometimes there is not.

It would be very useful to support components that generate _bulk_ telemetry:

1. They have a single function to call to write out an arbitrary telemetry channel, rather than hundreds of functions (one for each telemetry channel). This would be parameterized by the telemetry channel ID.
2. They would not specify the complete list of telemetry in FPP, but rather via some other mechanism. (Such as a spreadsheet or another autocoder.) This would get incorporated into the dictionary but not drive the FPP process.

## Rationale

Allows F Prime to be used in a wider range of applications without the need to manually "work around" the affordances of FPP.

timcanham, 2025-06-10:
This is a good idea - I had proposed something along these lines a while ago. Have a "nogen" tag for a channel in FPP that allows the ground system to decode it but allows the user to call a common function to bypass the codegen.

thomas-bc, 2025-06-10:
Just a few thoughts about your bullet point (2) :

> They would not specify the complete list of telemetry in FPP, but rather via some other mechanism. (Such as a spreadsheet or another autocoder.) 

Is there an inherent issue with using FPP-only? You could always have a `channels.fppi` that would be thousands of lines long, just like a spreadsheet would, and have FPP understand the notion of bulk channels. Could even autocode spreadsheet into that `channels.fppi` if needed.

> This would get incorporated into the dictionary but not drive the FPP process.

As I'm writing this I see Tim's comment above. Is the idea that users would provide that bulk write function? I'd think integrating it into FPP, maybe with a `bulk` tag instead of `nogen` would be beneficial ?

The reason I'm asking is that this all reminds me a lot of existing processes at JPL. Maybe there's better approaches in the context of F´. Maybe not.

timcanham, 2025-06-10:
My idea was this (couldn't find the git issue):

1) User chooses the channels in FPP to not generate by adding the tag
2) FPP doesn't generate the `tlmWrite...` functions, but provides a protected function something like:
`writeTlm(ID,buffer,time)`
3) User can the do:
```c++
for (id = 1 to x) {
    val = getTlm(x)
    writeTlm(x,val,now)
}
```
It could use a table, etc.

celskeggs, 2025-06-10:
@thomas-bc Does the current build system allow for autocoding other formats into FPP as part of the build process? Or would using another autocoder require running an additional manual step?

celskeggs, 2025-06-11:
@thomas-bc I'm aware that we can generate other kinds of code. Regarding FPP: The example you show uses `execute_process`. Is that the only way to generate FPP? That kind of solution seems a little hacky... I would expect it to result in performance problems in the build and generally not be as robust (for example, not regenerating when inputs are changed).

It might be a good way to get things done in the near term, but a sign that a more robust solution is needed in the long term. Unless there's a neater solution already available?