nasa/fprime #3732: Base ID clash not detect during build

jwest115, 2025-06-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

The following builds successfully when I would have expected an error to be thrown by FPP since the same base ID is used for each of the instances:
```
 instance spacePacketDeframer: Svc.CCSDS.SpacePacketDeframer base id 0x5000

 instance linuxTimer: Svc.LinuxTimer base id 0x5000
```

However, while I do not get an error when building on my machine, I see that CI does fail (in some cases):

Example CI failure (expected): https://github.com/nasa/fprime/actions/runs/15593763502/job/43919047783

Example CI pass (not expected): https://github.com/nasa/fprime/actions/runs/15591719918/job/43912175806

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Darwin
CPU Architecture: arm64
Platform: macOS-15.5-arm64-arm-64bit
Python version: 3.9.21
CMake version: 3.24.2
Pip version: 24.3.1
Pip packages:
    fprime-tools==4.0.0a1
    fprime-gds==4.0.0a3
    fprime-fpp-*==3.0.0a11
Project submodules:
```

Note: @thomas-bc updated the base IDs so they don't collide (see https://github.com/nasa/fprime/pull/3726), so I am running with an fprime version from the commit before the update.

A description of the expected behavior.

bocchino, 2025-06-12:
Item 3.c: "For each instance, this procedure creates a range of identifiers, from the base identifier to the largest identifier associated with the instance. If the component has no identifiers, then this range is empty."

The identifier range is the interval that goes from the smallest to the largest identifier associated with an instance I. If I has no identifiers, this range is empty. You could say that the identifier range for I contains just the base ID for I in that case, but then you'd be reserving one identifier (the base ID of I) that can never be used by any component instance, since I doesn't use it.

The intent is that if you have a lot of components with no dictionary identifiers, you don't have to reserve a different base ID for each one. This can matter on very small systems.

bocchino, 2025-06-12:
> The identifier range is the interval that goes from the smallest to the largest identifier associated with an instance I, including both endpoints.

This assumes that the component-local identifiers start at zero, which is usually the case. If the component-local identifiers don't start at zero, and there are identifiers, the system reserves the identifiers starting from the base ID.

bocchino, 2025-06-12:
That's true, according to the wording of the spec that shouldn't be allowed. I will investigate.

bocchino, 2025-06-13:
OK, it looks like I can reproduce the behavior by making the base IDs the same and swapping the order of the instances. I'll open an issue in the FPP repo.

bocchino, 2025-06-13:
Here is the FPP issue: https://github.com/nasa/fpp/issues/741. I think you can close this one as a duplicate.

closed 2025-06-13 by jwest115