nasa/fprime #3770: ByteStreamDriverModel should support synchronous sending

weggert2, 2025-06-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

For ip interfaces using synchronous sends, we would like for send calls to return a status, so that we can react to failures. To distinguish this from the generic `send` port in FPrime, We propose a syndSync port.

## Rationale

To use blocking / synchronous sends and get a status back, allowing calling code to retry, report an error, or other useful things.

LeStarch, 2025-06-20:
Recommendation:
- Return the implementation of the `send` port to be defined as `guarded` and return a status indicating success/fail.  We can handle async send calls when we have a real driver.

Willmac16, 2025-07-08:
Hi @LeStarch, I'm taking over this issue for @weggert2.

How do you want the existing `Drv::ByteStreamStatus` that is sent with the buffer through `sendReturnOut` to be handled?

* Should the `ByteStreamData` port be modified to not return a status?
* Should the guarded send call return the status in addition to `sendReturnOut`?

LeStarch, 2025-08-01:
We'd like to do this for 4.0 (next week).  Assigned @thomas-bc to work it.  Sorry for the confusion.

thomas-bc, 2025-08-04:
We will go ahead with a ByteStreamDriver (sync) and an AsyncByteStreamDriver interfaces. And ComStub should work with either.