nasa/fprime #3790: DecodingException import scope causes secondary failures

Aderiani, 2025-06-24:
| | |
|:---|:---|
|**_F´ Version_3.6.1| |
|**_Affected Component_3.6.1|   |
---
## Problem Description

In the fprime-gds, when event parsing fails, the exception handling itself fails with import errors, preventing graceful error recovery.

**Error**: `name 'DecodingException' is not defined`
**File**: `fprime_gds/common/distributor/distributor.py`

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.28.3
Pip version: 25.1.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==2.3.0
```

## How to Reproduce

1. Connect GDS to embedded system that sends events with different structure sizes or do not exist/mismatches in the dictionary that the gds is using
2. Check the logs for: distributor: Decoding error occurred: name 'DecodingException' is not defined. Skipping.

## Expected Behavior

Graceful error recovery

kevin-f-ortega, 2025-07-18:
@Aderiani, could you please provide a stack trace?

Aderiani, 2025-07-24:
> [@Aderiani](https://github.com/Aderiani), could you please provide a stack trace?

This is not treated as an Error by the gds code, therefore it does not throw any stacktrace. It appears as a warning if the log-to-stdout flag is set. The warning message is: [WARNING] distributor: Decoding error occurred: name 'DecodingException' is not defined. Skipping.