nasa/fprime #3832: Additional fprime-util check --coverage file creation

csmith608, 2025-07-01:
| | |
|:---|:---|
|**_F´ Version_**| 4.0.0a |
|**_Affected Component_**| Unit test framework |
---
## Feature Description

The files that are created when issuing a `fprime-util check --coverage` command include a summary of the coverage, as well as html files that show what is covered and what is not. Projects use this output as delivery products, but they also tend to want the output to the console shown when the unit test runs as a product too. Currently, we have to make these by hand, piping the output in to a test.txt file and move it in to the coverage folder. I propose that fprime automatically logs the output to terminal in a text file within the coverage folder. 

## Rationale

I have had multiple projects want this additional test data logged and I think fprime should do it automatically.

LeStarch, 2025-07-03:
I believe @timcanham suggested the same thing yesterday.

csmith608, 2025-07-03:
> I believe [@timcanham](https://github.com/timcanham) suggested the same thing yesterday.

He was in the meeting where this came up :)