nasa/fprime #3833: Add `Os::Atomic<T>` to the OSAL layer

LeStarch, 2025-07-01:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Currently, `std::atomic` is used as a way to implement atomic variables in F´.  However, `std::atomic` is not supported by all platforms.

Specifically, Atmel ATMega and ARM Cortex 6M (PICO 1) boards do not support `std::atomic`. 

Adding Atomic to the OSAL layer allows these platforms to provide alternate implementations rather than fail to compile F´.

Steps:
1. Enumerate functions of atomic variables used in F Prime (set, add, swap_if, etc).
2. Design OSAL interface
3. Implement OSAL interface via `std::atomic`

celskeggs, 2025-07-07:
My understanding was that `std::atomic` is mandatory for all C++ implementations, even freestanding implementations. See https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2023/n4950.pdf, Table 27. Am I misunderstanding? How are the Atmel ATMega and ARM Cortex 6M platforms standards-compliant if they don't support `std::atomic`?

bocchino, 2025-07-07:
An alternative approach is to provide our own implementation of `std::atomic`, or a relevant subset: https://en.cppreference.com/w/cpp/atomic/atomic.html. It's not clear to me why the non-compliant platforms aren't already doing this.

m-aleem, 2025-07-31:
The following Pull Requests are associated with this issue:

- [ ] 1) fprime-zephyr: https://github.com/fprime-community/fprime-zephyr/pull/11: Zephyr spinlock based std::atomic header-only implementation (+ updates for fprime-4.0.0 compatibility)
- [ ] 2) fprime-zephyr-reference: https://github.com/fprime-community/fprime-zephyr-reference/pull/4: Updates project CMakeLists.txt to use the Zephyr spinlock based std::atomic implementation from above PR for "rpi_pico" type boards (+ updates for fprime-4.0.0 compatibility)
- [ ] 3) fprime-baremetal: https://github.com/fprime-community/fprime-baremetal/pull/21 Implementation of a 'stub' std::atomic header-only implementation which is suitable for single-thread projects (note that this PR is opened against the fprime-baremetal branch already set up for fprime-4.0.0 compatibility)

closed 2025-08-19 by thomas-bc