nasa/fprime #3849: Fpy stack-based virtual machine

zimri-leisher, 2025-07-05:
| | |
|:---|:---|
|**_F´ Version_**| 4.0.0|
|**_Affected Component_**| FpySequencer |
---
## Feature Description

Transition the FpySequencer and Fpy compiler from register-based to stack-based.

## Rationale

Simpler to implement at compiler and virtual machine level. Register-based machines are often used because they're faster when the real hardware is register based, but in this case, the VM is executing at a much higher level than the actual hardware, so I doubt we get any performance gains. In any case, it's a scripting language, it doesn't need to be high performing. Stack-based machines are simpler to understand, write and compile code for.

zimri-leisher, 2025-07-08:
@bocchino  I was thinking more about non-constant sized lvars/non-word-aligned stacks, do you know if there are any languages with bytecodes which do this?

LeStarch, 2025-08-01:
@zimri-leisher why not use one of the project configurable sizes, like `FwSizeType`?