nasa/fprime #3854: FPrime GDS - Provide a Base URL/ Root URL option so that its easy to reverse proxy FPrime GDS

ephraim71, 2025-07-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| GDS |

## Feature Request: Configurable Base URL for GDS

Currently, FPrime's Ground Data System (GDS) can only run at a root URL (e.g., `localhost:5000`). This makes managing multiple FPrime instances challenging, requiring users to remember and track individual port numbers.

It would be helpful if GDS allowed specifying an arbitrary base URL (e.g., `localhost/fprime-gds-2`). This capability would simplify management of multiple GDS instances and facilitate use behind reverse proxies, similar to [Grafana's root URL option](https://grafana.com/tutorials/run-grafana-behind-a-proxy/).

kevin-f-ortega, 2025-07-18:
@thomas-bc @LeStarch , could you weigh in on this?

LeStarch, 2025-07-23:
I'd approve this from the CCB, with a note that this is not a feature we'd likely get to very soon.