nasa/fprime #3868: Topology visualization is broken for Ref

bocchino, 2025-07-09:
Running `fprime-util visualize` in `Ref` produces the following output:
```
Built target refresh_cache
fpp-to-layout
/Users/bocchino/JPL/Tools/FPrime/bocchino/fprime/Svc/Subtopologies/ComCcsds/ComCcsds.fpp:124.5
    topology Subtopology {
    ^
error: duplicate topology layout directory SubtopologyLayout
previous topology layout directory would be generated here:
/Users/bocchino/JPL/Tools/FPrime/bocchino/fprime/Svc/Subtopologies/CdhCore/CdhCore.fpp:51.5
    topology Subtopology {
    ^
```
In the current version of the visualizer tool chain, each topology has to have a different unqualified name.

Possible fixes:
1. Revise the visualizer tool chain to use fully-qualified topology names.
2. Revise the unqualified names in `Ref` so that they don't conflict.

Solution 1 is more robust, but solution 2 seems easier.

I am on this commit:
```
commit 6349bf85c23d589dd067767d2d19412b73af064b
Author: Justine West <35715959+jwest115@users.noreply.github.com>
Date:   Tue Jul 8 10:51:43 2025 -0700

    Add newline before table (#3858)
```

bocchino, 2025-07-16:
After discussion, let's go with solution 1 (fully-qualified names), since it seems to be a desired capability to pull subtopologies from several modules into a single visualization. When working across modules like this, we should use fully-qualified names.

jwest115, 2025-07-23:
Done in https://github.com/nasa/fpp/pull/760

closed 2025-07-23 by jwest115