nasa/fprime #3872: Type mismatch in FPP results in "undefined symbol" error

celskeggs, 2025-07-10:
| | |
|:---|:---|
|**_F´ Version_**| v4.0.0a1 |
|**_Affected Component_**| fpp |
---
## Problem Description

When compiling broken code like the following:

```
    enum ExampleEnum {
        # ...
    }
    passive component ExampleComponent {
        # ...
        sync input port examplePort: ExampleEnum
    }
```

FPP gives this error:

```
[build] fpp-to-cpp
[build] /workspaces/[...]/ExampleComponent.fpp:56.37
[build]         sync input port examplePort: ExampleEnum
[build]                                     ^
[build] error: undefined symbol ExampleEnum
```

This error message is misleading because the symbol _is_ defined... it's just the totally wrong type. An enum was specified instead of a port, and that's what's wrong, not a typo in the name.

## Expected Behavior

The error message should identify that the problem is a type mismatch.

I didn't realize that it was a type mismatch when I first saw this error, because I assumed that the "undefined symbol" error was caused by https://github.com/nasa/fprime/issues/2999. It wasn't until I did a full rebuild and it still failed that I realized I'd forgotten to define the port I wanted.

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.18.0-553.58.1.el8_10.x86_64-x86_64-with-glibc2.28
Python version: 3.9.20
CMake version: 3.24.2
Pip version: 25.1.1
Pip packages:
    fprime-tools==4.0.0a1
    fprime-gds==3.6.2a2.dev14+gabd041b
    fprime-fpp-*==3.0.0a10
Project submodules:
    https://github.com/nasa/fprime.git @ v4.0.0a1-1-g8e64db70e
    https://github.com/fprime-community/fprime-baremetal @ ca90b58
```

Kronos3, 2025-07-10:
This is due to the fact that ports and types have different namespaces (namegroups) in FPP. Technically you could define a port symbol and a type with the same name.

Would it be helpful to make the error more specific such as `undefined port symbol ExampleEnum`?

bocchino, 2025-07-10:
> This error message is misleading because the symbol is defined... it's just the totally wrong type. 

I see how this may be a natural way to think about what is going on, but as Andrei said it isn't how FPP works. In FPP there are several kinds of symbols, and it is permitted to define different symbols of different kinds with the same name. So if the system is looking for a port symbol S with name N, and no such S is defined, but an enum symbol S' exists with name N, then it's not accurate to say S "is defined with the wrong type." Instead, an unrelated symbol S' is defined, and it happens to have the same name. This design allows flexibility in naming. For example, it allows F Prime to have a type `Fw.Time` and a port `Fw.Time`.

To address the concern, we could have the error checker search all symbol kinds for unrelated symbols with the same name, but this seems excessive. Maybe a middle ground is to add an annotation of the form "note: a port symbol was expected here"? Would that be sufficient? That would be easy to add.

Also: the confusion in this case seems to be caused in part by #2999, which is a build system bug that we need to fix in any event.

celskeggs, 2025-07-11:
C++, Python, and most other languages that F Prime users are likely to know do not have separate namespaces for separate types. (C does have separate struct/union namespaces, but since these are denoted at every usage site, it's not as confusing.) I can see that the error message makes sense to the designers of the FPP language, but that doesn't mean it's the best choice for FPP's users.

> To address the concern, we could have the error checker search all symbol kinds for unrelated symbols with the same name, but this seems excessive.

Why is this excessive? Helpful diagnostics, including for "silly" mistakes like this one, should be [a first-class feature of modern languages and modern toolchains](https://clang.llvm.org/diagnostics.html). If a desirable diagnostic is difficult to implement due to the FPP compiler's design, then that seems like a flaw in the compiler design. Perhaps that's a good reason to deprioritize this request in favor of other requests that are more easily implemented, and come back to it when more time is available, but I don't think it's a good reason to descope the diagnostic permanently.

bocchino, 2025-07-11:
> Why is this excessive?

It seems sufficient to me to (1) report that a symbol is missing, and (2) to report the kind of symbol expected. Currently we are doing (1), and I agree it would be better to do (2) as well.

> Helpful diagnostics, including for "silly" mistakes like this one, should be [a first-class feature of modern languages and modern toolchains](https://clang.llvm.org/diagnostics.html).

Yes, this is true. The question is what's desirable. More isn't necessarily better. It seems to me that if we fix #2999 and report the kind of symbol that's missing, that will head off the mistake clearly and without excessive verbosity. If we also report the existence of unrelated symbols with the same name, I'm not entirely sure how to do that in a helpful way. For example, "note: the enum definition P is not a port definition" -- isn't this obvious? Do you have a specific suggestion for a more detailed error message?

celskeggs, 2025-07-16:
My suggestion for a more detailed error message:

```
[build] fpp-to-cpp
[build] /workspaces/[...]/ExampleComponent.fpp:56.37
[build]         sync input port examplePort: ExampleEnum
[build]                                     ^
[build] error: undefined port symbol ExampleEnum
[build] note: cannot use type symbol ExampleEnum here
```