nasa/fprime #3887: FpySequencer raise events in sequence

zimri-leisher, 2025-07-16:
| | |
|:---|:---|
|**_F´ Version_**| 4.0.0 |
|**_Affected Component_**| FpySequencer, fpyc |
---
## Feature Description

Add the ability to raise any event in a sequence.
This may be useful in some circumstances. However, it also might be an abuse of events, so maybe this isn't worth doing. It's technically possible though!

timcanham, 2025-07-16:
We usually used `CMD_NO_OP_STRING` to accomplish the same thing in the past, but you could have a sequencer directive with a string argument that could emit an event from Fpy.

zimri-leisher, 2025-07-17:
yeah, no op string would be fine, but I just realized that fprime's design would allow me to emit any event from any component from a sequence. It seems like a bad idea to do this but just kinda cool. Maybe I can have a LOG directive or something

closed 2025-08-01 by LeStarch