nasa/fprime #3891: Unable to run top level unit tests with Ninja

zimri-leisher, 2025-07-17:
| | |
|:---|:---|
|**_F´ Version_**| 4.0.0 52412392bebe50ee9a62df1344ff98f0dabbae5d |
|**_Affected Component_**| fprime-util check |
---
## Problem Description

In the top level dir of our repo, we usually run fprime-util generate --ut and then fprime-util check.
This runs all of the UTs for each component.
However, with the latest FPrime on devel and all the latest tools, running fprime-util check produces the following output:

[image]

This happens whether we use ninja or cmake.

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.26.0
Pip version: 22.0.2
Pip packages:
    fprime-tools==4.0.0a6
    fprime-gds==4.0.0a7
    fprime-fpp-*==3.0.0a13
Project submodules:
    https://github.com/nasa/fprime.git @ v4.0.0a1-100-g52412392b
```

vincewoo, 2025-07-21:
I also noticed this started happening sometime early last week. If you run `fprime-util check --all` it will run all the tests from the top-level directory. I've switched to doing that in the mean time.

LeStarch, 2025-07-22:
The fix was merged into devel

bocchino, 2025-07-22:
Reopening this issue. I'm seeing the same issue in FppTest after the fix.

bocchino, 2025-07-22:
@LeStarch  Let's avoid merging any PRs into devel until this is fixed. Code could be silently rotting in FppTest.