nasa/fprime #3913: Do not install static libraries to output directory

celskeggs, 2025-07-23:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**| build system |
---
## Feature Description

The current build system automatically installs static libraries like `libFw_Buffer.a` and `libSvc_Version.a` to the build-artifacts directory. I suggest disabling this.

## Rationale

While it would be valuable to install shared libraries, static libraries are already linked into the final executable, so the static libraries are unnecessary. My project uploads the build-artifacts folder to publish the compiled deployments as part of our CI process, and these library uploads waste space on the server.

LeStarch, 2025-07-23:
@timcanham would you weigh-in?  Many years ago you requested this feature to ensure that large projects could always have traceability, even through the archives.

I can see several possible resolutions:
1. Take this out, static archives *are* part of the final binary
2. Have it opt-in
3. Have it opt-out
4. Leave as always-on

I've always preferred item 1, but the system was designed as item 4 per @timcanham's request.

Note: shared libraries must always be installed, this only affects static libraries.

celskeggs, 2025-07-23:
I think this kind of traceability is better achieved by archiving the build directory, since there's a lot of information that doesn't get copied to `build-artifacts`.

LeStarch, 2025-08-01:
@SterlingPeet recommends this at least be an option.

SterlingPeet, 2025-08-01:
As is, it seems less useful, with headers there are some use cases that I have seen in-use around vendor shipments.

I do separately see value in rebuilding FSW for an in-progress mission, where the risk posture is dictating re-linking the new executable against the existing static libraries for which source was not adjusted.

celskeggs, 2025-08-01:
Given a real world use case (not just a hypothetical), I think it would be fine to make this configurable.

I'm personally skeptical of those two use cases:

1. I hope we don't get into the habit of encouraging vendors to provide binary-only F Prime libraries. It would be challenging to guarantee binary compatibility, and it would risk locking teams into using a specific version of F Prime in order to use a specific vendor's libraries. We should encourage vendors to release their source code instead.
2. For an in-progress mission, I'd rather we put effort into making sure our builds are [fully reproducible](https://reproducible-builds.org/), in which case we can guarantee bit-for-bit compatibility without the need for re-linking procedures.

celskeggs, 2025-08-05:
Goodness. That first scenario you talk about certainly sounds frustrating. I have a hard time understanding why the vendor organization would be legally allowed to provide you with the binary code but not with the source code... but I accept your point that this is a genuine real world use case that F Prime should accommodate. That would motivate the suggestion that the headers should be shipped as well.

I suggest that we add a flag to "opt in" to installing static libraries to the build-artifacts directory, instead of the current behavior of always installing them. I suggest "opt in" instead of "opt out" because the real world use cases described seem likely to occur less than 50% of the time in general.

I would also propose filing a separate ticket to install headers, which could be controlled by the same flag.

thomas-bc, 2025-08-05:
Agreed on the "opt in" flag proposal. This will also conveniently reduce the wall of "Installed file: <...>" logs at the end of each build.