nasa/fprime #3942: `fprime-util build --ut` runs the build over and over

bocchino, 2025-07-27:
For example, running `fprime-build --ut` in `Fw/Log` runs the unit test build ten times:
```
% fprime-util build --ut 2>&1 | grep 'Built target Fw_Log_ut_exe' | wc -l
10
```
In `Fw/Types`, the system appears to run the unit test build 20 times.

thomas-bc, 2025-07-29:
I'm not able to reproduce - can you share more about your setup and version ? How was the build cache generated ? Looks like this is using Makefiles which is not the default anymore

bocchino, 2025-07-29:
You are right, I configured with `fprime-util generate -f --make`. I have been doing that to work around this issue:

https://github.com/nasa/fprime/issues/3818

I'm running on Mac OS. I am on this commit: 1fe6a7cb046c2f443e57a685edac22281b4523f1. When I cd into `Fw/Types` and do a ut build, so everything is up-to-date and then run `fprime-util build --ut`, it runs the empty build over and over, taking 45 seconds to complete.

bocchino, 2025-07-29:
The ninja-based build does behave better, but unless we are decommissioning the make build, we should fix the make behavior.

LeStarch, 2025-08-01:
I am not able to reproduce this.  However, I will note that there is a subtle change in behavior:

1. If you `register` multiple things in a directory, fprime-util will invoke a targeted build of each
2. This implies you will get 2 builds when building UTs, one for the module and one for the test.

This is to ensure that the user can actually build everything in a directory and not just the first registered target.  You can build a specific target with the `--target` flag, which will build one unit only.

LeStarch, 2025-08-01:
Alright I did reproduce.  Will fix.

PR #3972 (2025-08-04): Bump fprime-tools version
| | |
|:---|:---|
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

FIXES #3952
FIXES #3942
closed 2025-08-04 by thomas-bc