nasa/fprime #3944: After FPP updates to new v4.0 serialization methods, mark legacy serialization methods as DEPRECATED

vincewoo, 2025-07-28:
This ticket is a follow-up to #3781 

As part of the changes to that ticket we chose not to DEPRECATE the now legacy serialization methods, serialize and deserialize. This was done so that we could decouple the change needed to FPP.

Once the FPP ticket (https://github.com/nasa/fpp/issues/763) is closed we should DEPRECATE the legacy methods in fprime to encourage the use of the new style.

vincewoo, 2025-08-01:
As part of this change we will have to undo some of the backwards compatibility workarounds:

- [ ]  Make deserializeFrom and serializeTo in `Fw::Serializable` pure virtual again and remove the default implementations. https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Types/Serializable.cpp#L20-L35
- [ ] Remove fallback logic of serializeFrom and deserializeTo in `Fw::SerializeBufferBase`. https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Types/Serializable.cpp#L269-L280 https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Types/Serializable.cpp#L562-L572 
- [ ] Remove fallback logic of serializeTo and deserializeFrom in `Fw::Com::ComPacket`. https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Com/ComPacket.cpp#L29-L37
- [ ] Remove definitions of serialization methods in `Fw::Com::ComPacket` and inherit from `Fw::Serializable` directly. https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Com/ComPacket.hpp#L27-L33
- [ ] Remove fallback logic of serializeTo and deserializeFrom in `Fw::Time`. https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Time/Time.cpp#L78-L101
- [ ] Remove SerializableStatus `FW_SERIALIZE_UNIMPLEMENTED` and `FW_DESERIALIZE_UNIMPLEMENTED` introduced by #3962

LeStarch, 2025-08-01:
> As part of this change we will have to undo the backwards compatibility workarounds that were introduced to decouple the FPP changes:
> 
>     * [ ]   Make deserializeFrom and serializeTo in `Fw::Serializable` pure virtual again and remove the default implementations. 
>         
>           
>             [fprime/Fw/Types/Serializable.cpp](https://github.com/nasa/fprime/blob/67be6defa09159da639285cd6b51db61b231f5d3/Fw/Types/Serializable.cpp#L20-L35)
>           
>           
>               Lines 20 to 35
>             in
>             [67be6de](/nasa/fprime/commit/67be6defa09159da639285cd6b51db61b231f5d3)
>           
>         
>         
>           
>       
>               
>                 
>                  // ---------------------------------------------------------------------- 
>               
>       
>               
>                 
>                  // Default implementations for new serialization methods 
>               
>       
>               
>                 
>                  // 
>               
>       
>               
>                 
>                  // These are provided for backward compatibility specifically for autocoding that 
>               
>       
>               
>                 
>                  // does not yet support the new serialization methods. 
>               
>       
>               
>                 
>                  // ---------------------------------------------------------------------- 
>               
>       
>               
>                 
>                   
>               
>       
>               
>                 
>                  SerializeStatus Serializable::serializeTo(SerializeBufferBase& buffer) const { 
>               
>       
>               
>                 
>                      // Default implementation for base class - derived classes should override this method 
>               
>       
>               
>                 
>                      return FW_SERIALIZE_FORMAT_ERROR; 
>               
>       
>               
>                 
>                  } 
>               
[... 401 lines cut ...]
>               
>                 
>                          status = this->m_val.deserialize(buffer); 
>               
>       
>               
>                 
>                      } 
>               
>       
>               
>                 
>                      return status; 
>               
>       
>               
>                 
>                  } 
>               
>           
>         
>       
>       [ ]  Remove SerializableStatus `FW_SERIALIZE_UNIMPLEMENTED` and `FW_DESERIALIZE_UNIMPLEMENTED` introduced by [Improve backward compatibility [de]serialize[From/To] fallback on Fw::Serializable and Fw::Buffer #3962](https://github.com/nasa/fprime/pull/3962)

To make sure I am tracking, this means we will have 4 functions: `serialize`, `serializeTo`, `deserialize`, and `deserializeFrom` that are all pure-virtual in the baseclass.  Does this imply that any implementor of a serializable will have to implement all 4?  Or will we have a base implementation of `serialize` and `deserialize` that calls `serializeTo` and `serializeFrom` thus limiting implementations to writing a single pair of functions.

vincewoo, 2025-08-06:
The clean-up has been merged in #3971 to get into 4.0.0. We will wait until 4.1.0 timeframe to perform the remaining deprecation.

vincewoo, 2025-09-22:
All legacy methods have been marked DEPRECATED in #4145