nasa/fprime #3983: Add Abilty to Turn Off Command Dispatcher Command Registration Events

LeStarch, 2025-08-05:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The command dispatcher emits many `DIAGNOSTIC` events when registering commands.  Add a configuration parameter to turn this behavior on/off.

## Rationale

This change is useful for memory-constrained systems where the size of downlink buffers must be restricted. If a deployment contains a sufficient number of commands (which can be exacerbated by the auto-generated parameter commands) the opcode registration causes a large number of diagnostic events which can fill the downlink buffer and crowd out more useful initialization events that occur later on. This change adds a simple compile-time flag which can toggle these events off and clean up the initialization event stream.

ianbrault, 2025-08-14:
This can be closed in favor of https://github.com/nasa/fprime/issues/4027

thomas-bc, 2025-08-15:
Closing per user request

closed 2025-08-15 by thomas-bc