nasa/fprime #4031: Upgrade to C++14

bocchino, 2025-08-14:
We should upgrade from C++11 to C++14 if we can. C++14 is basically C++11 with bug fixes.

LeStarch, 2025-08-15:
In order to approve this, we need @timcanham to test the `vxworks7` builds and we need to hear from @SterlingPeet if this is still an issue with atmega builds.

SterlingPeet, 2025-08-15:
It looks like Arduino is still shipping ``avr-gcc`` version 7.3 (updated from the last time we had this conversation, but not recently) and that version uses C++14 by default and can support up to C++17 when the ``-std`` flag is used.

My cursory glance indicates that other interesting arduino platforms are using newer version of GCC, so this should be OK from the perspective of building for ATmega.

LeStarch, 2025-08-15:
@SterlingPeet thanks for the feedback!  @kevin-f-ortega and @timcanham you are next.

kevin-f-ortega, 2025-08-27:
I compiled the PolarFire SoC VxWorks7 reference project with F Prime using C++14. It nominally ran on hardware without issue.

LeStarch, 2025-08-28:
@kevin-f-ortega what about the BeagleBone?

timcanham, 2025-08-29:
I was able to compile on our VxWorks 23.09 without problems.

kevin-f-ortega, 2025-08-29:
For posterity sake, this is how I compiled my deployments using C++14

I changed this line https://github.com/nasa/fprime/blob/v4.0.0/cmake/settings.cmake#L17 to 
```
    set(CMAKE_CXX_STANDARD 14)
```

This assumes my project doesn't define `CMAKE_CXX_STANDARD`.