nasa/fprime #4092: Move Data Product CRC calculation out of component thread context

kubiak-jpl, 2025-08-29:
| | |
|:---|:---|
|**_F´ Version_**| 4.0 |
|**_Affected Component_**| DataProducts |
---
## Feature Description

Move Data Product CRC calculation out of `dpSend`

## Rationale

While writing a record to a Data Product is generally a low overhead processing, sending the container to the DpWriter is surprisingly expensive, as the act of calling `dpSend` calls `container.updateDataHash()`. This will block the active thread while the hash is calculated, which can be an expensive task depending on the size of the data product. For data products of about 10 MB, this is enough to break my realtime loop that needs to run every 5ms.

Moving the CRC calculation to DpWriter would reduce the strain on the task which is writing to the DataProduct files and may have realtime requirements.

LeStarch, 2025-08-29:
I like this idea.  Is there a better place to put it?

kubiak-jpl, 2025-09-02:
For reference, pushing the `dpSend` calls to an internal thread was enough to work around the problem for what I needed

```cpp
  class SignalProcessing :
    public SignalProcessingComponentBase
  {
...
      // DpSend Thread
      std::jthread dpsend_thread_;
      std::deque<DpContainer> dpsend_q_;
      std::mutex dpsend_lock_;
      std::condition_variable dpsend_cv_;

  };

  void SignalProcessing ::
    dpsend_to_thread(
      DpContainer& c
  ) {

    std::unique_lock lk(dpsend_lock_);
    dpsend_q_.push_back(c);
    dpsend_cv_.notify_all();
  }

  void SignalProcessing ::
    dpsend_thread()
  {
    std::stop_token stop_token = dpsend_thread_.get_stop_token();
    while (true) {
      std::optional<DpContainer> container;
      {
        std::unique_lock lk(dpsend_lock_);
        dpsend_cv_.wait(lk);
        if (dpsend_q_.size() > 0) {
          container = dpsend_q_[0];
          dpsend_q_.pop_front();
        }
      }
      if (stop_token.stop_requested()) {
        break;
      }
      if (container) {
        dpSend(*container);
      }
    }
  }
```

closed 2026-04-15 by thomas-bc